orbbec/OrbbecSDK_ROS1 #5: Orbec Femto Mega Will not Stream 4K color resolution.

akashjinandra, 2023-09-12:
I have a femto mega that's running on a x86 based PC running Ubuntu 20 and ROS noetic. 

I am trying to run the color resolution at 3840x2160 at 25 FPS, and depth at NFOV unbinned 640x576 at 30 FPS.

I get this error:  
[ INFO] [1694559907.307522283]:  Failed to setup devices: Property is not supported! propertyId: 24
[ INFO] [1694559907.307522283]:  stream depth is enabled - width: 640, height: 576, fps: 30, Format: Y16
[ INFO] [1694559907.317091244]:  stream color is enabled - width: 3840, height: 2160, fps: 25, Format: MJPG
[ERROR] [1694559907.321991290]: failed to start pipeline: Current profile is not support sync config! try to disable ir stream try again
[ WARN] [1694559907.326653913]: Failed to start device: Current profile is not support sync config!

akashjinandra, 2023-09-13:
If I disable color or disable the depth then it works, but still throws this error: [ERROR] [1694563973.035897062]: Failed to setup devices: Property is not supported! propertyId: 24

akashjinandra, 2023-09-13:
fixed this by changing the depth FPS to match the color FPS to 25. Still not sure what the property error is.

hzcyf, 2023-09-20:
> fixed this by changing the depth FPS to match the color FPS to 25. Still not sure what the property error is.

You can ignore this property error directly, since the Femto Mega does not support this property in the new version.

closed 2023-09-25 by akashjinandra