orbbec/OrbbecSDK_ROS1 #6: Access Femto Mega through ethernet interface

akashjinandra, 2023-09-13:
I have a femto mega that's running on a x86 based PC running Ubuntu 20 and ROS noetic.

 I haven't been able to find any documentation on how to access or talk to the Femto Mega through the ethernet interface, Is there a default IP or settings to change in the ROS driver?

jian-dong, 2023-09-16:
@akashjinandra Current, I did not add ethernet access for ROS; I think the next release will be added. cc @zhonghong322

akashjinandra, 2023-10-05:
Is there any more effort on this? We would like to switch from the USB interface.

chengshans, 2024-08-08:
I encountered the same problem when using the Femto Mega I camera, is there any countermeasure now? Thanks.

akashjinandra, 2024-08-09:
Yes this is fully working in the latest release

closed 2024-08-09 by akashjinandra