orbbec/OrbbecSDK_ROS1 #9: GPU/computer requirements?

jeff-o, 2023-09-18:
Hello! Thanks so much for creating this library. I am planning out a vision system that could use up to three Orbbec Femto Mega cameras. Can you give me an idea of what sort of system requirements I would need to run such a system? Basic CPU/RAM/GPU numbers should be enough for me to plan. Thanks very much!

zhonghong322, 2023-09-20:
Which streams need to be opened for each of Femto Mega device, and what are the corresponding resolutions for each stream? 
The system configuration is different for different resolutions.

This also depends on the application scenario, such as the algorithm you have running. Can you introduce your application scenario?

jeff-o, 2023-09-20:
Thanks for your reply! Unfortunately it's tough to predict what the cameras will be used for. The system is being built for a client who is keeping the actual use-case private.

So I suppose it would be best to assume that they will be using maximum resolution, and heavy use of the depth sensing features (detailed 3D point cloud, etc). With that in mind, is it possible to suggest specs of a suitable host computer, or computers?

obshenlanzhou, 2025-09-08:
We will go ahead and close this issue. Please feel free to reopen it if you encounter any related problems.

closed 2025-09-08 by obshenlanzhou