orbbec/OrbbecSDK_ROS1 #16: Transforms is not valid in femto bolt.

Tacha-S, 2023-11-09:
If it were similar to Azure Kinect, `camera_link` and `color_frame` should be parallel, with `depth_frame` slightly pitched downward relative to `camera_link`. However, when starting FEMTO BOLT, the TF results in `camera_link` and `depth_frame` being parallel, with `color_frame` slightly pitched downward relative to `camera_link`.

Tacha-S, 2023-11-10:
Could you share CAD or URDF of femto bolt?

bmegli, 2024-01-15:
@Tacha-S, what I see in current master (54ca6bcb54ccc731879d4ea155dd9836de8c62f6) looks reasonable 

[image]

Adding `camera_link`
- parallel to RGB
- seems to share origin with `camera_depth_frame`
- `camera_depth_frame` is tilted down as in Kinect4A

[image]

bmegli, 2024-01-15:
This seems to be the relevant commit:
- https://github.com/orbbec/OrbbecSDK_ROS1/commit/95a2ce690408d08cd530b13045cd71570762ee41

Looks a bit overly optimistic
- assuming zero rotation for color
- assuming zero translation for depth

But maybe this is assumption in factory calibration

bmegli, 2024-01-15:
## Depth

```bash
rosrun tf tf_echo camera_link camera_depth_frame
At time 1705344616.484
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.004, 0.053, -0.001, 0.999]
            in RPY (radian) [-0.009, 0.107, -0.003]
            in RPY (degree) [-0.514, 6.105, -0.176]
```

## Color

```bash
rosrun tf tf_echo camera_link camera_color_frame
At time 1705344666.276
- Translation: [-0.006, -0.032, -0.001]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```

Tacha-S, 2024-01-16:
Thank you for the information. It has been corrected, so I will close this.

closed 2024-01-16 by Tacha-S