orbbec/OrbbecSDK_ROS1 #18: How could I get the IMU data

Ultramarine1939-syujie, 2023-11-23:
I've bought a gemini2,and The dealer told me that it has a built-in IMU, but it is not clear how to obtain the IMU data through ROS.
I'm working on ubuntu20.04-noetic

arjunskumar, 2024-02-26:
@Ultramarine1939-syujie have you managed to get IMU data?

arjunskumar, 2024-02-26:
@Ultramarine1939-syujie You have to modify [launch file](https://github.com/orbbec/OrbbecSDK_ROS1/blob/master/launch/gemini2_nodelet.launch)

```
<arg name="enable_accel" default="true"/>
<arg name="enable_gyro" default="true"/>
<arg name ="enable_sync_output_accel_gyro" default = "true" />
..
..
..

<param name="enable_sync_output_accel_gyro" value="$(arg enable_sync_output_accel_gyro)"/>

```

which will publish synchronized data stream of acceleration and gyroscope to the topic `camera/gyro_accel/sample`

obshenlanzhou, 2025-09-08:
We will go ahead and close this issue. Please feel free to reopen it if you encounter any related problems.