orbbec/OrbbecSDK_ROS1 #19: Use image transport to publish images

Tacha-S, 2023-11-27:
This node is trying to use `image_transport`, 
https://github.com/orbbec/OrbbecSDK_ROS1/blob/2385c916dcb5c1a3eae1891f5981bc20d638687f/CMakeLists.txt#L51

but it is actually a `ros::Publisher`
https://github.com/orbbec/OrbbecSDK_ROS1/blob/2385c916dcb5c1a3eae1891f5981bc20d638687f/include/orbbec_camera/ob_camera_node.h#L276

bmegli, 2024-01-18:
This is rather must-have in ROS (1 and 2)
- http://wiki.ros.org/image_transport

> image_transport should always be used to subscribe to and publish images. It provides transparent support for transporting images in low-bandwidth compressed formats.

Direct `Image` publishing is generally not used in ROS, it is handled by image_transport.

@Tacha-S changes for image_transport support adapted for current master can be found here:
- https://github.com/Extend-Robotics/OrbbecSDK_ROS1/commit/1ad41b525ff3890853915ea97aa3001eda06a267

bmegli, 2024-01-18:
A quick workaround if somebody needs to use image_transport without modifying code is to run `image_transport republish`

http://wiki.ros.org/image_transport (section 5)

Tacha-S, 2024-03-18:
fixed by https://github.com/orbbec/OrbbecSDK_ROS1/commit/38d8403ac2d0895dd7585af6343f216d8d530f86

closed 2024-03-18 by Tacha-S