orbbec/OrbbecSDK_ROS1 #22: missing install rule for orbbec_camera_node

bmegli, 2024-01-15:
The `CMakeLists.txt` currently doesn't install `orbbec_camera_node`

It installs `orbbec_camera_nodelet` only.

bmegli, 2024-01-15:
Maybe original intention was to
- fill in `EXECUTABLES` variable inside `add_orbbec_executable`

But it doesn't work as expected

https://github.com/orbbec/OrbbecSDK_ROS1/blob/54ca6bcb54ccc731879d4ea155dd9836de8c62f6/CMakeLists.txt#L216-L234

bmegli, 2024-01-15:
A quick workaround is to add  `${PROJECT_NAME}_node` it to install rules, e.g.

```CMake
# Install
install(TARGETS ${PROJECT_NAME} ${PROJECT_NAME}_node ${PROJECT_NAME}_nodelet ${EXECUTABLES}
  ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

BryanStuurman, 2024-02-28:
I have a PR that implements @bmegli 's theory that EXECUTABLES is to grow inside the add_obbec_executable macro, this PR is linked above.

bmegli, 2024-03-18:
Fixed in 28f0813

closed 2024-03-18 by bmegli