orbbec/OrbbecSDK_ROS1 #24: Issues with running multiple (4) orbbec cameras - gemini2

dhruvah, 2024-01-24:
I'm running into issues while running four cameras. I have a file that launches gemini2.launch, with the right namespaces, and a delay between the launches. All the camera nodes come up, and the logs show that everything is connected. I have all the serial numbers set correctly, so that is not an 

But when I echo the topic for the fourth camera, it doesn't seem to publish any information.

This behavior is also inconsistent, and sometimes all the four cameras launch correctly with all the data being streamed.

I suspect that the error could be how the streaming is handled, based on the Subscriber callbacks. But I am not sure.

Has anyone run into a similar issue? This is my python launch file

```

#!/usr/bin/env python

import roslaunch
import rospy

rospy.init_node('multi_camera_launcher')
uuid = roslaunch.rlutil.get_or_generate_uuid(None, False)

launch = roslaunch.scriptapi.ROSLaunch()
launch.start()

front_up_depthcam = rospy.get_param('~front_up_depthcam', True)
back_up_depthcam = rospy.get_param('~back_up_depthcam', True)
front_down_depthcam = rospy.get_param('~front_down_depthcam', True)
back_down_depthcam = rospy.get_param('~back_down_depthcam', True)

device_num = rospy.get_param('~device_num', 4)

front_up_serial = rospy.get_param('~front_up_serial', '')
back_up_serial = rospy.get_param('~back_up_serial', '')
front_down_serial = rospy.get_param('~front_down_serial', '')
back_down_serial = rospy.get_param('~back_down_serial', '')

launch_file_path = '../src/orbbec_sdk_ros1/launch/gemini2.launch'

time_buffer = 3

def launch_camera(namespace, serial, device_num):
    cli_args = [
        launch_file_path,
        'camera_name:=' + namespace,
        'serial_number:=' + serial,
        'device_num:=' + str(device_num)
    ]
    roslaunch_args = cli_args[1:]
    roslaunch_file = [(roslaunch.rlutil.resolve_launch_arguments(cli_args)[0], roslaunch_args)]
    launch.parent = roslaunch.parent.ROSLaunchParent(uuid, roslaunch_file)
    launch.start()
    
 if front_up_depthcam:
    launch_camera('front_up', front_up_serial, device_num)

rospy.sleep(time_buffer)

if back_up_depthcam:
    launch_camera('back_up', back_up_serial, device_num)

rospy.sleep(time_buffer)

if front_down_depthcam:
    launch_camera('front_down', front_down_serial, device_num)

rospy.sleep(time_buffer)

if back_down_depthcam:
    launch_camera('back_down', back_down_serial, device_num)
    
launch.spin()
```

dhruvah, 2024-01-25:
What seems to be an issue is that for the camera that is not working correctly `pipeline_->start()` in ob_camera_node.cpp doesn't seem to be working when `startStreams()` is called

dhruvah, 2024-01-31:
The funny thing is that, this works perfectly fine for 3 cameras.

It has something to do when you start subscribing using a fourth camera, that it just doesn't work

alonks1234, 2025-05-05:
https://www.orbbec.com/docs/g330-troubleshooting/

See question 9 - likely a usb buffer size issue that kicks in at 4 cameras.

Try:

```
sudo nano /etc/default/grub

# Find the line that starts # GRUB_CMDLINE_LINUX_DEFAULT= and replace it with:
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.autosuspend=-1 usbcore.lpm_enable=0 usbcore.old_scheme_first=Y usbcore.usbfs_memory_mb=128"

# Update grub & reboot
sudo update-grub
sudo reboot
```

obshenlanzhou, 2025-09-08:
Hi @dhruvah Is this issue resolved? If you don’t need further assistance, we will go ahead and close this issue. Please feel free to reopen it if you encounter any related problems.

closed 2025-09-08 by obshenlanzhou