orbbec/OrbbecSDK_ROS1 #25: [ERROR] get gain error Property is not supported! propertyId: 2018/2027

yuukireina05, 2024-01-26:
你好，我正在使用设备Femto Bolt并在我的ROS系统上成功运行了这个ROS驱动。
运行是成功的，也能通过RVIZ可视化点云，但是在log里面出现了两个错误
```shell
[ERROR] [1706263552.921097452]: get gain error Property is not supported! propertyId: 2018
[ERROR] [1706263552.921116392]: get gain error Property is not supported! propertyId: 2027
``` 
想询问一下产生这个错误的原因是什么？
完整的log如下
```shell
started roslaunch server http://xiaofeng-lab-workstation:37959/

SUMMARY
========

PARAMETERS
 * /camera/orbbec_camera/accel_range: 4g
 * /camera/orbbec_camera/accel_rate: 100hz
 * /camera/orbbec_camera/angular_vel_cov: 0.01
 * /camera/orbbec_camera/camera_name: camera
 * /camera/orbbec_camera/color_delay_us: 0
 * /camera/orbbec_camera/color_format: MJPG
 * /camera/orbbec_camera/color_fps: 25
 * /camera/orbbec_camera/color_height: 960
 * /camera/orbbec_camera/color_info_uri: 
 * /camera/orbbec_camera/color_width: 1280
 * /camera/orbbec_camera/connection_delay: 100
 * /camera/orbbec_camera/depth_delay_us: 0
 * /camera/orbbec_camera/depth_format: Y16
 * /camera/orbbec_camera/depth_fps: 25
 * /camera/orbbec_camera/depth_height: 576
 * /camera/orbbec_camera/depth_registration: True
 * /camera/orbbec_camera/depth_width: 640
 * /camera/orbbec_camera/device_num: 1
 * /camera/orbbec_camera/enable_accel: False
 * /camera/orbbec_camera/enable_color: True
 * /camera/orbbec_camera/enable_color_auto_exposure: True
 * /camera/orbbec_camera/enable_colored_point_cloud: True
 * /camera/orbbec_camera/enable_d2c_viewer: False
 * /camera/orbbec_camera/enable_depth: True
 * /camera/orbbec_camera/enable_frame_sync: False
 * /camera/orbbec_camera/enable_gyro: False
 * /camera/orbbec_camera/enable_ir: True
 * /camera/orbbec_camera/enable_ir_auto_exposure: True
 * /camera/orbbec_camera/enable_pipeline: True
 * /camera/orbbec_camera/enable_point_cloud: True
 * /camera/orbbec_camera/enable_soft_filter: True
 * /camera/orbbec_camera/enable_sync_output_accel_gyro: True
 * /camera/orbbec_camera/flip_color: False
 * /camera/orbbec_camera/flip_depth: False
 * /camera/orbbec_camera/flip_ir: False
 * /camera/orbbec_camera/gyro_range: 1000dps
 * /camera/orbbec_camera/gyro_rate: 100hz
 * /camera/orbbec_camera/ir_format: Y16
 * /camera/orbbec_camera/ir_fps: 25
 * /camera/orbbec_camera/ir_height: 576
 * /camera/orbbec_camera/ir_info_uri: 
 * /camera/orbbec_camera/ir_width: 640
 * /camera/orbbec_camera/liner_accel_cov: 0.01
 * /camera/orbbec_camera/log_level: none
 * /camera/orbbec_camera/product_id: 
 * /camera/orbbec_camera/publish_tf: True
 * /camera/orbbec_camera/serial_number: 
 * /camera/orbbec_camera/sync_mode: free_run
 * /camera/orbbec_camera/tf_publish_rate: 10.0
 * /camera/orbbec_camera/trigger2image_delay_us: 0
 * /camera/orbbec_camera/trigger_out_delay_us: 0
 * /camera/orbbec_camera/trigger_out_enabled: False
 * /camera/orbbec_camera/usb_port: 
 * /camera/orbbec_camera/vendor_id: 0x2bc5
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /camera/
    orbbec_camera (nodelet/nodelet)
    orbbec_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [261818]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 79cfa622-bc32-11ee-a275-175379641319
process[rosout-1]: started with pid [261859]
started core service [/rosout]
process[camera/orbbec_camera_manager-2]: started with pid [261866]
process[camera/orbbec_camera-3]: started with pid [261867]
[ INFO] [1706263547.730849858]: Loading nodelet /camera/orbbec_camera of type orbbec_camera/OBCameraNodelet to manager orbbec_camera_manager with the following remappings:
[ INFO] [1706263547.731332068]: /camera/depth/color/points -> /camera/depth_registered/points
[ INFO] [1706263547.732323548]: waitForService: Service [/camera/orbbec_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1706263547.737022496]: Initializing nodelet with 32 worker threads.
[ INFO] [1706263547.753135442]: waitForService: Service [/camera/orbbec_camera_manager/load_nodelet] is now available.
[01/26 19:05:47.761933][info][261866][Context.cpp:67] Context created with config: /home/xiaofeng/git/moonshot_robot_pc/catkin_ws/src/OrbbecSDK_ROS1/config/OrbbecSDKConfig_v1.0.xml
[01/26 19:05:47.761943][info][261866][Context.cpp:72] Context work_dir=/home/xiaofeng/.ros
[01/26 19:05:47.858672][info][261866][LinuxPal.cpp:112] Create PollingDeviceWatcher!
[01/26 19:05:47.858702][info][261866][DeviceManager.cpp:15] Current found device(s): (1)
[01/26 19:05:47.858709][info][261866][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8S93P000Y, Connection: USB3.1
[ INFO] [1706263547.861792244]: queryDevice: first query device
[ INFO] [1706263547.861830714]: deviceConnectCallback : deviceConnectCallback start
[ INFO] [1706263547.961929869]: deviceConnectCallback : Before process lock lock
[ INFO] [1706263547.961956379]: deviceConnectCallback : After process lock lock
[ INFO] [1706263547.961968499]: deviceConnectCallback : selectDevice start
[ INFO] [1706263547.961988119]: Connecting to the default device
[ INFO] [1706263548.831082156]: deviceConnectCallback : selectDevice end
[ INFO] [1706263548.853409761]:  stream depth is enabled - width: 640, height: 576, fps: 25, Format: Y16
[ INFO] [1706263548.853638160]:  stream ir is enabled - width: 640, height: 576, fps: 25, Format: Y16
[ INFO] [1706263548.854967910]:  stream color is enabled - width: 1280, height: 960, fps: 25, Format: MJPG
[ INFO] [1706263548.858058129]: set align mode:  ALIGN_D2C_SW_MODE
[ INFO] [1706263549.348437605]: Create color frame read thread.
[ WARN] [1706263552.907535865]: Publishing dynamic camera transforms (/tf) at 10 Hz
[ INFO] [1706263552.917484933]: Stream left_ir is disabled.
[ INFO] [1706263552.917503923]: Stream right_ir is disabled.
[ INFO] [1706263552.920600672]: stream depth exposure 125
[ INFO] [1706263552.920827992]: stream ir exposure 125
[ INFO] [1706263552.920993472]: stream color exposure 200
[ERROR] [1706263552.921097452]: get gain error Property is not supported! propertyId: 2018
[ERROR] [1706263552.921116392]: get gain error Property is not supported! propertyId: 2027
[ INFO] [1706263552.921282702]: stream color gain 0
[ INFO] [1706263552.921443372]: stream color wb 6500
[ INFO] [1706263552.921484632]: Device Femto Bolt connected
[ INFO] [1706263552.921495472]: Serial number: CL8S93P000Y
[ INFO] [1706263552.921504132]: Firmware version: 1.0.6
[ INFO] [1706263552.921512232]: Hardware version: 1.2
[ INFO] [1706263552.921518552]: device uid: 6-4-2
[ INFO] [1706263552.921526192]: deviceConnectCallback : deviceConnectCallback end
[ INFO] [1706263621.150552136]: rgb point cloud subscribed
[ INFO] [1706263621.150590487]: Image stream depth subscribed
[ INFO] [1706263621.150619347]: set align mode:  ALIGN_D2C_SW_MODE
[ INFO] [1706263621.297080832]: Image stream color subscribed
[ INFO] [1706263621.297112613]: pipe line already started
[ INFO] [1706263714.153176107]: point cloud unsubscribed
[ INFO] [1706263714.153212487]: Image stream depth unsubscribed
[ INFO] [1706263716.664619292]: Image stream color unsubscribed
[ INFO] [1706263716.664668983]: imageUnsubscribedCallback pipe line not start
^C[camera/orbbec_camera-3] killing on exit
[camera/orbbec_camera_manager-2] killing on exit
[ INFO] [1706263717.480986844]: Unloading nodelet /camera/orbbec_camera from manager orbbec_camera_manager
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

yuukireina05, 2024-01-26:
I am using Femto Bolt in a Ubuntu 20.04, ROS Noetic workstation. 
The driver's running was successful, but I found the error logs in the driver's output. 
I would like to know how to eliminate the errors. What may be the reasons causing the errors? 
I appreciate your help!

ignpaub, 2024-01-26:
Same by here. ROS noetic / Ubuntu 20.04 with a Femto Bolt camera

MidasOverbeeke, 2024-03-07:
Same here on ROS Melodic, please fix this soon as I need it for my current project.

MidasOverbeeke, 2024-03-07:
> @ignpaub The log level is not set properly, please just ignore this error.

But I still do not receive any messages on /camera/depth/points topic, how can I fix this?

jian-dong, 2024-03-07:
> > @ignpaub The log level is not set properly, please just ignore this error.
> 
> But I still do not receive any messages on /camera/depth/points topic, how can I fix this?

Do you have changed the fixed frame to `camera_link` in rviz?

MidasOverbeeke, 2024-03-07:
> > > @ignpaub The log level is not set properly, please just ignore this error.
> > 
> > 
> > But I still do not receive any messages on /camera/depth/points topic, how can I fix this?
> 
> Do you have changed the fixed frame to `camera_link` in rviz?

Yes, but this is not the problem. When I launch femto_bolt.launch and run this is a seconds terminal:
rostopic echo /camera/depth/points I don't see any messages. I should see a PointCloud2 message every 1/30th of a second, right?

jian-dong, 2024-03-07:
> Yes, but this is not the problem. When I launch femto_bolt.launch and run this is a seconds terminal: rostopic echo /camera/depth/points I don't see any messages. I should see a PointCloud2 message [...]

Yes, it should be like this, can you get the depth image in Rviz? And can you please paste
the log when camera start here, I need the firmware version， and `rostopic /camera/depth/camera_info` result, Thank you. my time here is 10:55 PM, I'll take care of this issue tomorrow.

MidasOverbeeke, 2024-03-07:
`Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:43625/

SUMMARY
========

PARAMETERS
 * /camera/camera/accel_range: 4g
 * /camera/camera/accel_rate: 100hz
 * /camera/camera/angular_vel_cov: 0.01
 * /camera/camera/camera_name: camera
 * /camera/camera/color_delay_us: 0
 * /camera/camera/color_format: MJPG
 * /camera/camera/color_fps: 30
 * /camera/camera/color_height: 720
 * /camera/camera/color_info_uri: 
 * /camera/camera/color_width: 1280
 * /camera/camera/connection_delay: 100
 * /camera/camera/depth_delay_us: 0
 * /camera/camera/depth_format: Y16
 * /camera/camera/depth_fps: 30
 * /camera/camera/depth_height: 576
 * /camera/camera/depth_registration: False
 * /camera/camera/depth_width: 640
 * /camera/camera/device_num: 1
 * /camera/camera/enable_accel: False
 * /camera/camera/enable_color: True
 * /camera/camera/enable_color_auto_exposure: True
 * /camera/camera/enable_colored_point_cloud: True
 * /camera/camera/enable_d2c_viewer: False
 * /camera/camera/enable_depth: True
 * /camera/camera/enable_frame_sync: False
 * /camera/camera/enable_gyro: False
 * /camera/camera/enable_ir: True
 * /camera/camera/enable_ir_auto_exposure: True
 * /camera/camera/enable_pipeline: True
 * /camera/camera/enable_point_cloud: True
 * /camera/camera/enable_soft_filter: True
 * /camera/camera/flip_color: False
 * /camera/camera/flip_depth: False
 * /camera/camera/flip_ir: False
 * /camera/camera/gyro_range: 1000dps
 * /camera/camera/gyro_rate: 100hz
 * /camera/camera/ir_format: Y16
 * /camera/camera/ir_fps: 30
 * /camera/camera/ir_height: 576
 * /camera/camera/ir_info_uri: 
 * /camera/camera/ir_width: 640
 * /camera/camera/liner_accel_cov: 0.01
 * /camera/camera/log_level: none
 * /camera/camera/product_id: 
 * /camera/camera/publish_tf: True
 * /camera/camera/serial_number: 
 * /camera/camera/sync_mode: free_run
 * /camera/camera/tf_publish_rate: 10.0
 * /camera/camera/trigger2image_delay_us: 0
 * /camera/camera/trigger_out_delay_us: 0
 * /camera/camera/trigger_out_enabled: False
 * /camera/camera/usb_port: 
 * /camera/camera/vendor_id: 0x2bc5
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /camera/
    camera (orbbec_camera/orbbec_camera_node)

auto-starting new master
process[master]: started with pid [27581]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 10dce3c4-dc93-11ee-9505-000c29023502
process[rosout-1]: started with pid [27592]
started core service [/rosout]
process[camera/camera-2]: started with pid [27599]
[03/07 15:57:50.349905][info][27599][Context.cpp:67] Context created with config: /home/student/catkin_ws/src/orbbec/OrbbecSDK_ROS1/config/OrbbecSDKConfig_v1.0.xml
[03/07 15:57:50.349929][info][27599][Context.cpp:72] Context work_dir=/home/student/.ros
[03/07 15:57:50.455944][info][27599][LinuxPal.cpp:110] Create PollingDeviceWatcher!
[03/07 15:57:50.455995][info][27599][DeviceManager.cpp:15] Current found device(s): (1)
[03/07 15:57:50.456055][info][27599][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL000000000, Connection: USB3.1
[ INFO] [1709823470.459117428]: queryDevice: first query device
[ INFO] [1709823470.459728831]: deviceConnectCallback : deviceConnectCallback start
[ INFO] [1709823470.560829076]: deviceConnectCallback : Before process lock lock
[ INFO] [1709823470.560876007]: deviceConnectCallback : After process lock lock
[ INFO] [1709823470.560884265]: deviceConnectCallback : selectDevice start
[ INFO] [1709823470.560893007]: Connecting to the default device
[ INFO] [1709823471.228907039]: deviceConnectCallback : selectDevice end
[ERROR] [1709823471.268982484]: Failed to setup devices: Property is not supported! propertyId: 24
[ INFO] [1709823471.269178137]:  stream depth is enabled - width: 640, height: 576, fps: 30, Format: Y16
[ INFO] [1709823471.269250153]:  stream ir is enabled - width: 640, height: 576, fps: 30, Format: Y16
[ INFO] [1709823471.270125519]:  stream color is enabled - width: 1280, height: 720, fps: 30, Format: MJPG
[ INFO] [1709823471.274455017]: set align mode:  ALIGN_D2C_SW_MODE
[ INFO] [1709823472.278359540]: Create color frame read thread.
[ WARN] [1709823475.439335496]: Publishing dynamic camera transforms (/tf) at 10 Hz
[ INFO] [1709823475.452815380]: Stream left_ir is disabled.
[ INFO] [1709823475.452876519]: Stream right_ir is disabled.
[ INFO] [1709823475.459701603]: stream depth exposure 125
[ INFO] [1709823475.460362298]: stream ir exposure 125
[ INFO] [1709823475.460797004]: stream color exposure 200
[ERROR] [1709823475.460877506]: get gain error Property is not supported! propertyId: 2018
[ERROR] [1709823475.460937117]: get gain error Property is not supported! propertyId: 2027
[ INFO] [1709823475.461292510]: stream color gain 0
[ INFO] [1709823475.461642780]: stream color wb 6500
[ INFO] [1709823475.461736697]: Device Femto Bolt connected
[ INFO] [1709823475.461785515]: Serial number: CL8FC3100JN
[ INFO] [1709823475.461820996]: Firmware version: 1.0.9
[ INFO] [1709823475.461848331]: Hardware version: 1.2
[ INFO] [1709823475.461874199]: device uid: 4-1-4
[ INFO] [1709823475.461900754]: deviceConnectCallback : deviceConnectCallback end`

rostopic echo /camera/depth/camera_info does nothing. Nothing is pubished there, but the topic is available (so a pubisher is enabled, but does nothing?)

zhonghong322, 2024-03-08:
Does the depth data appear correctly when testing with orbbecviewer?

jian-dong, 2024-03-08:
Hi @MidasOverbeeke,

Unfortunately, I was unable to reproduce the issue on my computer. Could you please use the OrbbecViewer to check if the depth image and point cloud are working properly? You can download it from the following link: [OrbbecViewer v1.9.5](https://github.com/orbbec/OrbbecSDK/releases/download/v1.9.5/OrbbecViewer_v1.9.5_202403050503_linux_x64_release.zip).

Thank you for your cooperation.

MidasOverbeeke, 2024-03-08:
> Hi @MidasOverbeeke,
>
> Unfortunately, I was unable to reproduce the issue on my computer. Could you please use the OrbbecViewer to check if the depth image and point cloud are working properly? You [...]

So this is what I did what I tried to run it, but am I doing something wrong? I have to say I'm not great with Linux.
[image]
It seems like it doesn't run, do I need to change permissions somewhere? I've made everything executable with chmod +x.

jian-dong, 2024-03-08:
@MidasOverbeeke, could you please run `uname -a` and share the result here? It seems that your computer is not running on x64 Arch.

MidasOverbeeke, 2024-03-08:
> @MidasOverbeeke, could you please run `uname -a` and share the result here? It seems that your computer is not running on x64 Arch.

[image]

jian-dong, 2024-03-08:
> ![image](https://private-user-images.githubusercontent.com/149528353/311197938-435b14e8-e09e-4066-8dde-a4afb797f529.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]
The issue might be due to the file being uncompressed on Windows before being copied to Linux. I recommend decompressing the zip file directly on Linux instead. This method could potentially avoid the problems you're experiencing. Please give it a try and let me know how it goes.  CC @zhonghong322

zhonghong322, 2024-03-08:
If you extract the files on Windows and then copy them to Linux, the dynamic libraries may lose symbolic links, and you will need to extract them on Linux instead.

MidasOverbeeke, 2024-03-08:
> If you extract the files on Windows and then copy them to Linux, the dynamic libraries may lose symbolic links, and you will need to extract them on Linux instead.

Yes, that was indeed the problem, but I still don't see any depthpoints.
[image]
I see an error `Depth engine create and initialize failed,retCode:204`, is that the problem maybe?

This is all the output I got:
```2024-03-08 04:09:16.203 ERROR [1869] [CamTool::logger::init@111] OBViewerLog config failed
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@18] **********************************************
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@19]  OrbbecViewer launched! Welcome!! 
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@20]  	- Version: V1.9.5
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@21]  	- Author: 
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@22]  	- E-Mail: 
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@23]  	- Company: orbbec
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@24]  	- Website: http://www.orbbec.com.cn/
2024-03-08 04:09:16.203 INFO  [1869] [loggerInit@25] **********************************************
[03/08 04:09:16.733822][info][1869][Context.cpp:69] Context created with config: default config!
[03/08 04:09:16.733845][info][1869][Context.cpp:74] Context work_dir=/home/student
[03/08 04:09:16.733849][info][1869][Context.cpp:77] 	- SDK version: 1.9.5
[03/08 04:09:16.733881][info][1869][Context.cpp:78] 	- SDK stage version: main
[03/08 04:09:16.733898][info][1869][Context.cpp:82] get config EnumerateNetDevice:false
[03/08 04:09:16.733918][info][1869][LinuxPal.cpp:38] createObPal: create LinuxPal!
[03/08 04:09:16.836758][info][1869][LinuxPal.cpp:112] Create PollingDeviceWatcher!
[03/08 04:09:16.836910][info][1869][DeviceManager.cpp:15] Current found device(s): (1)
[03/08 04:09:16.836930][info][1869][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC3100JN, Connection: USB3.1
[03/08 04:09:16.837095][info][1869][DeviceManager.cpp:310] Enable net device enumeration: true
[03/08 04:09:16.837615][info][1869][GVCPClient.cpp:223] bind 0.0.0.0:0
[03/08 04:09:17.839276][info][1869][DeviceManager.cpp:15] Current device(s) list: (1)
[03/08 04:09:17.839298][info][1869][DeviceManager.cpp:24] 	- Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC3100JN, Connection: USB3.1
[03/08 04:09:17.839975][info][1869][FemtoBoltUvcDevice.cpp:23] FemtoBoltUvcDevice init ...
[03/08 04:09:17.840340][info][1869][FemtoBoltUvcDevice.cpp:120] Create command start!
[03/08 04:09:17.843974][info][1869][MSDEConverterDevice.cpp:721] Succeed to load depth engine plugin
[03/08 04:09:18.053151][info][1869][FemtoBoltUvcDevice.cpp:271] Create command done!
[03/08 04:09:18.053172][info][1869][FemtoBoltUvcDevice.cpp:431] init sensor map start!
[03/08 04:09:18.053179][info][1869][FemtoBoltUvcDevice.cpp:458] init sensor map done!
[03/08 04:09:18.054100][info][1869][AbstractDevice.cpp:124] 	- Firmware version: 1.0.9
[03/08 04:09:18.054182][info][1869][FemtoBoltUvcDevice.cpp:275] Init depth process param start!
[03/08 04:09:18.068315][info][1869][MSDEConverterDevice.cpp:772] got nvram data succeed.
[03/08 04:09:18.068344][info][1869][ObUvcDevice.cpp:112] endpoint:130
[03/08 04:09:18.069552][info][1869][ObUvcDevice.cpp:118] libusb_clear_halt done, endpoint:130
[03/08 04:09:18.529360][info][1869][FemtoBoltUvcDevice.cpp:402] setNvramDataStreamStopFunc succeed
[03/08 04:09:18.529436][info][1869][FemtoBoltUvcDevice.cpp:427] Init depth process param done!
[03/08 04:09:18.531994][info][1869][FemtoBoltUvcDevice.cpp:38] FemtoBoltUvcDevice init done!
[03/08 04:09:18.532021][info][1869][DeviceManager.cpp:157] Device created successfully! Name: Femto Bolt, PID: 0x066b, SN/ID: CL8FC3100JN
[03/08 04:09:18.534212][info][1869][FemtoBoltUvcDevice.cpp:619] Ir sensor has been created!
[03/08 04:09:18.536207][info][1869][FemtoBoltUvcDevice.cpp:585] Color sensor has been created!
[03/08 04:09:18.541200][info][1869][FemtoBoltUvcDevice.cpp:554] Depth sensor has been created!
[03/08 04:09:18.542233][error][1869][HandleLibusb.hpp:84] handle_libusb - detach kernel driver
[03/08 04:09:18.542322][error][1869][IMUFrameReversion.cpp:8] entry IMUFrameReversion
[03/08 04:09:18.542386][info][1869][AccelSensor.cpp:26] AccelSensor created
[03/08 04:09:18.542421][info][1869][FemtoBoltUvcDevice.cpp:681] Accel sensor has been created!
[03/08 04:09:18.545172][error][1869][IMUFrameReversion.cpp:8] entry IMUFrameReversion
[03/08 04:09:18.545209][info][1869][GyroSensor.cpp:27] GyroSensor created!
[03/08 04:09:18.545229][info][1869][FemtoBoltUvcDevice.cpp:723] Gyro sensor has been created!
[03/08 04:09:18.548634][info][1869][Pipeline.cpp:44] Pipeline created with device: {name: Femto Bolt, sn: CL8FC3100JN}, @0x23940F0
[03/08 04:09:18.554742][info][1905][DeviceManager.cpp:249] dev-uid@0x4-1-2, update device time succeeded! round-trip-time=1ms

2024-03-08 04:09:18.563 INFO  [1907] [CurlUtil::firmwareListGetCurl@187]  ?series=FemtoBolt
curl_easy_perform() failed: URL using bad/illegal format or missing URL
2024-03-08 04:09:18.564 ERROR [1907] [CurlUtil::firmwareListGetCurl@213] Response is null
2024-03-08 04:09:18.564 ERROR [1907] [UpgradeOnlineServicer::CheckFWVersion@28] Get firmwareList is Failed!
2024-03-08 04:09:18.564 ERROR [1907] [UpgradeOnlineServicer::OnlineUpgradeDetection@87] Check firmware of version Failed!
2024-03-08 04:09:20.535 INFO  [1892] [PropertyManager::syncProperty@341] property supported: 14@Mirror Depth Stream
[... cut ...]
[03/08 04:09:26.387478][error][1938][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[03/08 04:09:26.388029][info][1938][MSDEConverterDevice.cpp:75] Depth engine got nvram data size:490936
[03/08 04:09:26.388077][info][1938][MSDEConverterDevice.cpp:102] use dynlib load depthengine lib......
[03/08 04:09:26.472626][error][1938][MSDEConverterDevice.cpp:109] Depth engine create and initialize failed,retCode:204
[03/08 04:09:26.473189][info][1869][Pipeline.cpp:252] Start streams done!
[03/08 04:09:26.473207][info][1869][Pipeline.cpp:235] Pipeline start done!```

MidasOverbeeke, 2024-03-11:
Hi, I've been investigating the problem myself, and found that the depth camera device is not found, see this image:
[image]
What could be the problem here? It does show up on my Windows system, but VMware seems to not route the depth sensor to the Linux system... Do you use VMware too or do you run Linux natively?

bmegli, 2024-03-11:
> Hi, I've been investigating the problem myself, and found that the depth camera device is not found, see this image:

`2bc5:066b `is Femto Bolt, see Orbbec [udev rules](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L6)

Have you installed udev rules by the way?

What I see natively is (and that is all related to FemtoBolt):

```bash
[USER]@[HOST]:~$ lsusb
Bus 002 Device 004: ID 2bc5:066b Orbbec Orbbec Femto Bolt 3D Camera
```

--------------------------------------------

> As I have found [here](https://github.com/orbbec/OrbbecSDK/blob/main/doc/tutorial/English/Environment_Configuration.md), the results should include these two devices:

`2bc5:0536` is Astra+ RGB, see [here](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L9)

`2bc5:0636` is Astra+, see [here](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L7)

MidasOverbeeke, 2024-03-11:
> > Hi, I've been investigating the problem myself, and found that the depth camera device is not found, see this image:
> 
> `2bc5:066b `is Femto Bolt, see Orbbec [udev rules](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L6)
> 
> Have you installed udev rules by the way?
> 
> What I see natively is (and that is all related to FemtoBolt):
> 
> ```shell
> [USER]@[HOST]:~$ lsusb
> Bus 002 Device 004: ID 2bc5:066b Orbbec Orbbec Femto Bolt 3D Camera
> ```
> 
> > As I have found [here](https://github.com/orbbec/OrbbecSDK/blob/main/doc/tutorial/English/Environment_Configuration.md), the results should include these two devices:
> 
> `2bc5:0536` is Astra+ RGB, see [here](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L9)
> 
> `2bc5:0636` is Astra+, see [here](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L7)

Yes I have, and it seems like you're seeing the same device, so the UBS device and driver is the same. I think that this is a problem with depthengine, like you said.

zhonghong322, 2024-03-12:
> > > Hi, I've been investigating the problem myself, and found that the depth camera device is not found, see this image:
> > 
> > 
> > `2bc5:066b `is Femto Bolt, see Orbbec [udev rules](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L6)
> > Have you installed udev rules by the way?
> > What I see natively is (and that is all related to FemtoBolt):
> > ```shell
> > [USER]@[HOST]:~$ lsusb
> > Bus 002 Device 004: ID 2bc5:066b Orbbec Orbbec Femto Bolt 3D Camera
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > > As I have found [here](https://github.com/orbbec/OrbbecSDK/blob/main/doc/tutorial/English/Environment_Configuration.md), the results should include these two devices:
> > 
> > 
> > `2bc5:0536` is Astra+ RGB, see [here](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L9)
> > `2bc5:0636` is Astra+, see [here](https://github.com/orbbec/OrbbecSDK_ROS1/blob/f0abb4c1c45c5d3b9e6baac9908dc54cee1ea75c/script/99-obsensor-libusb.rules#L7)
> 
> Yes I have, and it seems like you're seeing the same device, so the USB device and driver is the same. I think that this is a problem with depthengine, like you said. Could it be that I'm running linux in a VMware instance, so no real graphics card is connected?

yes，Depth engine cannot be used on VMware.

obshenlanzhou, 2025-09-08:
If you don’t need further assistance, we will go ahead and close this issue. Please feel free to reopen it if you encounter any related problems.