orbbec/OrbbecSDK_ROS1 #35: Gemini 2XL crashes on v1.5.3

maximilianwulf, 2024-05-14:
When running `v1.5.3` with `roslaunch orbbec_camera gemini2XL.launch` the software crashes with :

```
terminate called after throwing an instance of 'libobsensor::wrong_api_call_sequence_exception'
  what():  Resource busy! You can try again later!
```

Orbbec viewer works out of the box, so I would assume it is an issue of this package.

Full logs: 

```
[05/14 17:36:33.633220][info][47465][Context.cpp:68] Context created with config: /home/username/config/OrbbecSDKConfig_v1.0.xml
[05/14 17:36:33.633235][info][47465][Context.cpp:73] Work directory=/home/username/.ros, SDK version=v1.10.5-20240511-7dc8be9
[05/14 17:36:33.633249][info][47465][LinuxPal.cpp:31] createObPal: create LinuxPal!
[05/14 17:36:33.759261][info][47465][LinuxPal.cpp:142] Create PollingDeviceWatcher!
[05/14 17:36:33.759332][info][47465][DeviceManager.cpp:15] Current found device(s): (1)
[05/14 17:36:33.759347][info][47465][DeviceManager.cpp:24] 	- Name: Orbbec Gemini2 XL, PID: 0x0671, SN/ID: AYMXB32000M, Connection: USB2.0
[ INFO] [1715700993.768352936]: queryDevice: first query device
[ INFO] [1715700993.769478816]: deviceConnectCallback : deviceConnectCallback start
[ INFO] [1715700993.869747970]: deviceConnectCallback : Before process lock lock
[ INFO] [1715700993.869829477]: deviceConnectCallback : After process lock lock
[ INFO] [1715700993.869853344]: deviceConnectCallback : selectDevice start
[ INFO] [1715700993.869878425]: Connecting to the default device
[ INFO] [1715700994.364419853]: deviceConnectCallback : selectDevice end
[ INFO] [1715700994.742692914]: Depth process is HW D2D
[ INFO] [1715700994.757602062]:  stream depth is enabled - width: 640, height: 400, fps: 10, Format: Y16
[ INFO] [1715700994.758299590]:  stream color is enabled - width: 640, height: 400, fps: 10, Format: MJPG
[ INFO] [1715700994.758552084]:  stream left_ir is enabled - width: 640, height: 400, fps: 10, Format: Y8
[ INFO] [1715700994.758795198]:  stream right_ir is enabled - width: 640, height: 400, fps: 10, Format: Y8
[ INFO] [1715700994.758838470]: stream gyro full scale range 1000dps sample rate 100hz
[ INFO] [1715700994.758874756]: stream accel full scale range 4g sample rate 100hz
[ INFO] [1715700994.803752743]: Publishing static transform from camera_link to color
[ INFO] [1715700994.803772690]: Translation 24.9924, -0.00599234, -0.208842
[ INFO] [1715700994.803782888]: Rotation 0.000570562, -2.71207e-05, 0.000561105, 1
[ INFO] [1715700994.803792333]: Publishing static transform from camera_link to depth
[ INFO] [1715700994.803800060]: Translation 0, 0, 0
[ INFO] [1715700994.803829204]: Rotation 0, 0, 0, 1
[ INFO] [1715700994.803836222]: Publishing static transform from camera_link to left_ir
[ INFO] [1715700994.803843975]: Translation 0, 0, 0
[ INFO] [1715700994.803850484]: Rotation 0, 0, 0, 1
[ INFO] [1715700994.803858411]: Publishing static transform from camera_link to right_ir
[ INFO] [1715700994.803868319]: Translation 100.49, 0, 0
[ INFO] [1715700994.803875641]: Rotation 0, 0, 0, 1
[ WARN] [1715700994.804128860]: Publishing dynamic camera transforms (/tf) at 10 Hz
[ INFO] [1715700994.805261041]: Stream ir is disabled.
[ INFO] [1715700994.811937538]: stream depth exposure 6000
[ INFO] [1715700994.812478367]: stream color exposure 6000
[ INFO] [1715700994.813205338]: stream left_ir exposure 6000
[ INFO] [1715700994.813823605]: stream right_ir exposure 6000
[ INFO] [1715700994.814192363]: stream depth gain 1000
[ INFO] [1715700994.814612390]: stream color gain 256
[ INFO] [1715700994.814981938]: stream left_ir gain 1000
[ INFO] [1715700994.815452419]: stream right_ir gain 1000
[ INFO] [1715700994.815909735]: stream color wb 5000
[ INFO] [1715700994.816032296]: Device Orbbec Gemini 2 XL connected
[ INFO] [1715700994.816046298]: Serial number: AYMXB32000M
[ INFO] [1715700994.816055403]: Firmware version: 2.1.3_23112010
[ INFO] [1715700994.816069616]: Hardware version: V1.0
[ INFO] [1715700994.816082521]: device uid: 1-7-15
[ INFO] [1715700994.816097969]: Current node pid: 47465
[ INFO] [1715700994.816105872]: deviceConnectCallback : deviceConnectCallback end
[ INFO] [1715701000.993639209]: Image stream depth subscribed
[ INFO] [1715701000.993820032]: set align mode to HW
terminate called after throwing an instance of 'libobsensor::wrong_api_call_sequence_exception'
  what():  Resource busy! You can try again later!
[camera/camera-2] process has died [pid 47465, exit code -6, 
```

zhonghong322, 2024-05-29:
We will fix this issue in the next version.

maximilianwulf, 2024-05-29:
Great, thanks a lot.

zhonghong322, 2024-06-18:
@maximilianwulf  This issue has been fixed. Please test it using the latest version.

maximilianwulf, 2024-06-18:
Thanks a lot.

maximilianwulf, 2024-06-20:
Can confirm it works. Thanks a lot!

closed 2024-06-20 by maximilianwulf