orbbec/OrbbecSDK_ROS1 #39: [camera/camera-2] process has died [pid xxxx, exit code -11 ...

zzy257248, 2024-05-22:
Hello, I'm running a femto-bolt camera under ROS-noetic, Ubuntu20.04. It works well in OrbbecViewer, but fails with terminal commands. I tried to start the camera and view the image in Rviz, but usually after I entered "roslaunch orbbec_camera femto_bolt.launch" I got error message like this:

[camera/camera-2] process has died [pid xxxx, exit code -11, cmd /home/xxx/xxx/devel/lib/orbbec_camera/orbbec_camera_node /camera/depth/color/points:=/camera/depth_registered/points __name:=camera __log:=/home/xxx/.ros/log/91bf7772-1852-11ef-bde4-4bb8f2ffcf85/camera-camera-2.log].
log file: /home/xxx/.ros/log/91bf7772-1852-11ef-bde4-4bb8f2ffcf85/camera-camera-2*.log

Sometimes I could successfully run that command by pulling the USB cable out and reinserting it, but still failed to view the image in Rviz, as Status Warn showing: "No Image Received". Strangely, I was able to view the image in Rviz this way days ago. The problem still exists after I upgraded OrbbecSDK_ROS1 to the newest version.

Looking forward to an answer! Many thanks!

jian-dong, 2024-05-29:
It seems like the issue might be related to your Femto Bolt camera's USB connection. The camera should be recognized as a USB 3.0 device, but it might be detected as USB 2.0, which can cause problems. Please check the logs to confirm the connection type.

To verify if the camera is recognized as USB 3.0, look for a log similar to this:

```plaintext
[05/29 17:47:33.922206][info][145503][LinuxPal.cpp:32] createObPal: create LinuxPal!
[05/29 17:47:33.925154][info][145503][LinuxPal.cpp:143] Create PollingDeviceWatcher!
[05/29 17:47:34.926162][info][145503][DeviceManager.cpp:15] Current found device(s): (1)
[05/29 17:47:34.926184][info][145503][DeviceManager.cpp:23]  - Name: Femto Bolt, PID: 0x066b, SN/ID: CL8P83D0005, Connection: USB3.1
```

zzy257248, 2024-06-03:
Thanks so much! It has been very helpful!

closed 2024-06-03 by zzy257248