orbbec/OrbbecSDK_ROS1 #42: multiple cameras - device_num logic unnecessary

bmegli, 2024-07-08:
Currently when using multiple sensors user has to pass number of sensors to driver
- this is the only place in code I can see `device_num` used

This is unnecessary and difficult for user (what if sensors are added dynamically to the system by user?)

It should be enough to:
- check if serial and USB descriptor are both empty (not defined)
  - connect to default sensor in that case
- connect by serial otherwise
- connect by USB descriptor otherwise

Link to relevant code fragment:

https://github.com/orbbec/OrbbecSDK_ROS1/blob/c587b26c7c0191af5006ec76a8a32945f307c324/src/ob_camera_node_driver.cpp#L92-L114

bmegli, 2024-07-08:
Example implementation in:
- https://github.com/Extend-Robotics/OrbbecSDK_ROS1/commit/07daf95ed28dd8c3afa1ddba6afd95ed84c92cea

bmegli, 2025-03-07:
The new implemented logic is now reasonable.

- try by serial if defined
- try by usb port if defined 
- try connecting to anything if num_devices==1

Num_devices only role remains to guard against not defined serial/usb_port if num_devices>0

https://github.com/orbbec/OrbbecSDK_ROS1/blob/dcaea6dbd520f145cc47468ccece3bbf75b4c657/src/ob_camera_node_driver.cpp#L265-L294

closed 2025-03-07 by bmegli

obshenlanzhou, 2025-09-08:
Hi @bmegli If you don’t need further assistance, we will go ahead and close this issue. Please feel free to reopen it if you encounter any related problems.