orbbec/OrbbecSDK_ROS1 #46: Femto Mega I support

bmegli, 2024-09-17:
Is Femto Mega I supported by this driver?

As I understand it only supports ethernet inteterface.

-----------------------------------------------------------------------------------

This issue:
- https://github.com/orbbec/OrbbecSDK_ROS1/issues/6
- suggests that it should be supported

Readme supported hardware:
- https://github.com/orbbec/OrbbecSDK_ROS1?tab=readme-ov-file#supported-hardware-products
  - guides to OrbbecSDK support
    - https://github.com/orbbec/OrbbecSDK?tab=readme-ov-file#product-support
    - which mentions Femto Mega
      - but doesn't mention Femto Mega I

------------------------------------------------------------------------------------------

Could you clarify support for Femto Mega I?

zhonghong322, 2024-10-08:
What are the firmware version and SDK version of your Mega I device?

bmegli, 2024-10-10:
Hi @zhonghong322,

This is theoretical question. 

Other Femto variants are listed as supported in docs:
- https://github.com/orbbec/OrbbecSDK?tab=readme-ov-file#product-support

Is Femto Mega I not supported or just not listed?

bmegli, 2024-11-28:
Clarified in 295d8ea593728a70388a9d557868c25054aeb107

closed 2024-11-28 by bmegli