orbbec/OrbbecSDK_ROS1 #49: Invalid Quaternion for Femto Bolt

marcomasa, 2024-09-25:
When enabling the "publish_tf" option, other nodes dependand on TF are logging errors:

```
[ERROR] [1727277411.522716528]: Ignoring transform for child_frame_id "femto_bolt_accel_frame" from authority "unknown_publisher" because of a nan value in the transform (-nan -0.000000 -inf) (-nan -nan -nan -nan)
[ERROR] [1727277411.522808829]: Ignoring transform for child_frame_id "femto_bolt_accel_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
[ERROR] [1727277411.522854529]: Ignoring transform for child_frame_id "femto_bolt_gyro_frame" from authority "unknown_publisher" because of a nan value in the transform (-nan -0.000000 -inf) (-nan -nan -nan -nan)
[ERROR] [1727277411.522912130]: Ignoring transform for child_frame_id "femto_bolt_gyro_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
```

The ROS message in /femto_bolt/gyro_accel/sample also shows an invalid quaternion (cannot be 4 times zero!) :

```
header: 
  seq: 249
  stamp: 
    secs: 1727277526
    nsecs: 640000000
  frame_id: "camera_gyro_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.001064225216396153
  y: -0.0026605629827827215
  z: 0.01489915233105421
angular_velocity_covariance: [0.0001, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0001]
linear_acceleration: 
  x: -0.013159179128706455
  y: -0.05861816182732582
  z: -9.943553924560547
linear_acceleration_covariance: [0.0001, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0001]
---
```

How can i solve this?

marcomasa, 2024-09-25:
This is also "confirmed" by the start logs:

```
[ INFO] [1727279518.445303048]: Image stream depth subscribed
[ INFO] [1727279518.610421738]: Publishing static transform from ir to depth
[ INFO] [1727279518.610572059]: Translation 0, 0, 0
[ INFO] [1727279518.610682975]: Rotation 0, 0, 0, 1
[ INFO] [1727279518.610719780]: Publishing static transform from color to depth
[ INFO] [1727279518.610818494]: Translation 32.387, 1.58323, -2.90842
[ INFO] [1727279518.610912207]: Rotation 0.00136692, -0.0531993, 0.00166793, 0.998582
[ INFO] [1727279518.611009320]: Publishing static transform from depth to depth
[ INFO] [1727279518.611105734]: Translation 0, 0, 0
[ INFO] [1727279518.611175343]: Rotation 0, 0, 0, 1
[ INFO] [1727279518.611280158]: Publishing static transform from accel to depth
[ INFO] [1727279518.611391173]: Translation 0, inf, -nan
[ INFO] [1727279518.611500589]: Rotation -nan, -nan, -nan, -nan
[ INFO] [1727279518.611610604]: Publishing static transform from gyro to depth
[ INFO] [1727279518.611723620]: Translation 0, inf, -nan
[ INFO] [1727279518.611830034]: Rotation -nan, -nan, -nan, -nan
```

How can i set this manually? Or is there even a way to do so?

marcomasa, 2024-09-25:
I added some custom debug prints after line 1668 in **ob_camera_node.cpp** while searching for the error:

```
// ob_camera_node.cpp
// line 1667 onwards 

    try {
      ex = stream_profile->getExtrinsicTo(base_stream_profile);

      // custom debugging code
      ROS_INFO_STREAM(log_prefix << "Extrinsics for " << stream_name_[stream_index]);
      int i = 0;
      for (auto& rot_elem : ex.rot) {
        ROS_INFO_STREAM(log_prefix << "Rot [ " << i++ << "] = [" << rot_elem << "]");
      }
      i = 0;
      for (auto& trans_elem : ex.trans) {
        ROS_INFO_STREAM(log_prefix << "Trans [ " << i++ << "] = [" << trans_elem << "]");
      }

    } catch (const ob::Error& e) {
```

It seems that the extrinsics of the gyro and accel stream are wrong:

```
[ INFO] [1727298240.384982681]: CUSTOM-DEBUG    Extrinsics for accel
[ INFO] [1727298240.385100784]: CUSTOM-DEBUG    Rot [ 0] = [0]
[ INFO] [1727298240.385224888]: CUSTOM-DEBUG    Rot [ 1] = [0]
[ INFO] [1727298240.385385792]: CUSTOM-DEBUG    Rot [ 2] = [0]
[ INFO] [1727298240.385488295]: CUSTOM-DEBUG    Rot [ 3] = [0]
[ INFO] [1727298240.385570697]: CUSTOM-DEBUG    Rot [ 4] = [0]
[ INFO] [1727298240.385701501]: CUSTOM-DEBUG    Rot [ 5] = [0]
[ INFO] [1727298240.385777203]: CUSTOM-DEBUG    Rot [ 6] = [-nan]
[ INFO] [1727298240.385883806]: CUSTOM-DEBUG    Rot [ 7] = [-inf]
[ INFO] [1727298240.385990108]: CUSTOM-DEBUG    Rot [ 8] = [-nan]
[ INFO] [1727298240.386117012]: CUSTOM-DEBUG    Trans [ 0] = [0]
[ INFO] [1727298240.386193314]: CUSTOM-DEBUG    Trans [ 1] = [inf]
[ INFO] [1727298240.386272316]: CUSTOM-DEBUG    Trans [ 2] = [-nan]
[ INFO] [1727298240.386358218]: Publishing static transform from accel to depth
[ INFO] [1727298240.386446421]: Translation 0, inf, -nan
[ INFO] [1727298240.386531723]: Rotation -nan, -nan, -nan, -nan
[ INFO] [1727298240.386667727]: CUSTOM-DEBUG    Extrinsics for gyro
[ INFO] [1727298240.386742929]: CUSTOM-DEBUG    Rot [ 0] = [0]
[ INFO] [1727298240.386827331]: CUSTOM-DEBUG    Rot [ 1] = [0]
[ INFO] [1727298240.386932534]: CUSTOM-DEBUG    Rot [ 2] = [0]
[ INFO] [1727298240.387020937]: CUSTOM-DEBUG    Rot [ 3] = [0]
[ INFO] [1727298240.387106239]: CUSTOM-DEBUG    Rot [ 4] = [0]
[ INFO] [1727298240.387190341]: CUSTOM-DEBUG    Rot [ 5] = [0]
[ INFO] [1727298240.387267543]: CUSTOM-DEBUG    Rot [ 6] = [-nan]
[ INFO] [1727298240.387345745]: CUSTOM-DEBUG    Rot [ 7] = [-inf]
[ INFO] [1727298240.387465749]: CUSTOM-DEBUG    Rot [ 8] = [-nan]
[ INFO] [1727298240.387577452]: CUSTOM-DEBUG    Trans [ 0] = [0]
[ INFO] [1727298240.387685555]: CUSTOM-DEBUG    Trans [ 1] = [inf]
[ INFO] [1727298240.387820858]: CUSTOM-DEBUG    Trans [ 2] = [-nan]
[ INFO] [1727298240.387914961]: Publishing static transform from gyro to depth
[ INFO] [1727298240.388010663]: Translation 0, inf, -nan
[ INFO] [1727298240.388104066]: Rotation -nan, -nan, -nan, -nan
```

However, i cannot trace further than to **StreamProfile.hpp** (Line 45):
```
    /**
     * @brief Get the extrinsic parameters from current stream profile to the given target stream profile
     *
     * @return OBExtrinsic Return the extrinsic parameters.
     */
    OBExtrinsic getExtrinsicTo(std::shared_ptr<StreamProfile> target);
```

Are the extrinsics loaded from the camera?
If yes, how can i set them ?

marcomasa, 2024-09-25:
I also checked my firmware and it should be the latest available one (1.0.9):

[image]

jian-dong, 2024-09-26:
Hi @marcomasa 
I updated the SDK and fixed this bug. Can you try it? My FW version is 1.1.2.

marcomasa, 2024-09-26:
Hi @jian-dong 

Thank you so much for your fast response!
I just pulled your fix and it seems to resolve the TF issue.
So far i have not gotten any warnings and my output now looks much better:

```
[ INFO] [1727351181.794123764]: CUSTOM-DEBUG    Extrinsics for accel
[ INFO] [1727351181.794228017]: CUSTOM-DEBUG    Rot [ 0] = [2.67949e-08]
[ INFO] [1727351181.794283639]: CUSTOM-DEBUG    Rot [ 1] = [-1]
[ INFO] [1727351181.794362128]: CUSTOM-DEBUG    Rot [ 2] = [-5.55112e-17]
[ INFO] [1727351181.794424141]: CUSTOM-DEBUG    Rot [ 3] = [0.104528]
[ INFO] [1727351181.794509221]: CUSTOM-DEBUG    Rot [ 4] = [2.80083e-09]
[ INFO] [1727351181.794591005]: CUSTOM-DEBUG    Rot [ 5] = [0.994522]
[ INFO] [1727351181.794644530]: CUSTOM-DEBUG    Rot [ 6] = [-0.994522]
[ INFO] [1727351181.794670992]: CUSTOM-DEBUG    Rot [ 7] = [-2.66481e-08]
[ INFO] [1727351181.794763961]: CUSTOM-DEBUG    Rot [ 8] = [0.104528]
[ INFO] [1727351181.794857230]: CUSTOM-DEBUG    Trans [ 0] = [-37.8292]
[ INFO] [1727351181.794924036]: CUSTOM-DEBUG    Trans [ 1] = [-1.29141]
[ INFO] [1727351181.794987546]: CUSTOM-DEBUG    Trans [ 2] = [54.174]
[ INFO] [1727351181.795049459]: Publishing static transform from accel to depth
[ INFO] [1727351181.795080215]: Translation -37.8292, -1.29141, 54.174
[ INFO] [1727351181.795177079]: Rotation 0.525483, 0.473147, -0.473147, 0.525483
[ INFO] [1727351181.795267751]: CUSTOM-DEBUG    Extrinsics for gyro
[ INFO] [1727351181.795332559]: CUSTOM-DEBUG    Rot [ 0] = [2.67949e-08]
[ INFO] [1727351181.795362218]: CUSTOM-DEBUG    Rot [ 1] = [-1]
[ INFO] [1727351181.795458282]: CUSTOM-DEBUG    Rot [ 2] = [-5.55112e-17]
[ INFO] [1727351181.795622151]: CUSTOM-DEBUG    Rot [ 3] = [0.104528]
[ INFO] [1727351181.795689456]: CUSTOM-DEBUG    Rot [ 4] = [2.80083e-09]
[ INFO] [1727351181.795716718]: CUSTOM-DEBUG    Rot [ 5] = [0.994522]
[ INFO] [1727351181.795814580]: CUSTOM-DEBUG    Rot [ 6] = [-0.994522]
[ INFO] [1727351181.795885679]: CUSTOM-DEBUG    Rot [ 7] = [-2.66481e-08]
[ INFO] [1727351181.795990432]: CUSTOM-DEBUG    Rot [ 8] = [0.104528]
[ INFO] [1727351181.796077409]: CUSTOM-DEBUG    Trans [ 0] = [-37.8292]
[ INFO] [1727351181.796130834]: CUSTOM-DEBUG    Trans [ 1] = [-1.29141]
[ INFO] [1727351181.796147410]: CUSTOM-DEBUG    Trans [ 2] = [54.174]
[ INFO] [1727351181.796174772]: Publishing static transform from gyro to depth
[ INFO] [1727351181.796251064]: Translation -37.8292, -1.29141, 54.174
[ INFO] [1727351181.796316072]: Rotation 0.525483, 0.473147, -0.473147, 0.525483
```

However, the IMU message still contains an invalid quaternion:
```
---
header: 
  seq: 359
  stamp: 
    secs: 1727351692
    nsecs: 768999936
  frame_id: "camera_gyro_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.003192675532773137
  y: 0.001064225216396153
  z: 0.014367040246725082
angular_velocity_covariance: [0.0001, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0001]
linear_acceleration: 
  x: 0.07895507663488388
  y: -0.02033691294491291
  z: -9.943553924560547
linear_acceleration_covariance: [0.0001, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.0001]
---
```

I hotfixed this by modifying line 825 in **setDefaultIMUMessage** in **ob_camera_node.cpp**:

```
void OBCameraNode::setDefaultIMUMessage(sensor_msgs::Imu& imu_msg) {
  imu_msg.header.frame_id = "imu_link";
  imu_msg.orientation.x = 0.0;
  imu_msg.orientation.y = 0.0;
  imu_msg.orientation.z = 0.0;
  // imu_msg.orientation.w = 0.0; causes invalid quaternion
  imu_msg.orientation.w = 1.0;

  imu_msg.orientation_covariance = {-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0};
  imu_msg.linear_acceleration_covariance = {
      liner_accel_cov_, 0.0, 0.0, 0.0, liner_accel_cov_, 0.0, 0.0, 0.0, liner_accel_cov_};
  imu_msg.angular_velocity_covariance = {
      angular_vel_cov_, 0.0, 0.0, 0.0, angular_vel_cov_, 0.0, 0.0, 0.0, angular_vel_cov_};
}
```

I am not sure if this is actually needed, since the Femto IMU doesn't support orientation, but i suspect that other packages might be thrown off by the invalid quaternion. So you might wanna think about integrating this aswell :)

jian-dong, 2024-09-26:
I believe what you said is correct, and I have added your fix in the latest commit.

commit 203ce480 (2024-09-26): Update SDK to v1.10.14-20240926-5a40ade for fix #49
closed 2024-09-26 by marcomasa