orbbec/OrbbecSDK_ROS1 #55: gemini335 camera launch error

xiaoye1994, 2024-10-29:
I used ubuntu20.04 based on raspberry 4B device  and  my camera is gemini 335 .
I have intalled the orbbec_camera packdge  successfully and I use 
"rosluanch orbbec_camera gemini_330_series.launch"  command to start the device .
I already got device id and it can be recognaized by the usb 3.0 port. But I got this error. What's the issue ,ths .

[image]

jian-dong, 2024-10-31:
can you paste the camera_carash log here

xiaoye1994, 2024-11-06:
> can you paste the camera_carash log here

now I have tried the code on my computer virtual machine , and it works well.  But my rasipberry 4B still can not run the code , I also checked the memory state and it dose not have any memory leak problem.
 My virtual machine uses OS version is ubuntu 18.04 ,and my rasipberry device uses ubuntu 20 version.  You think it will be the point to lead my camera crash or any other reasons ?

Srikiranrao, 2024-12-09:
@xiaoye1994  any update on this issue. 
I am facing the same issue.

closed 2025-10-30 by obshenlanzhou