orbbec/OrbbecSDK_ROS1 #57: Depth image stream fails with this error message

siwook-choi, 2024-11-01:
After initialization, depth image is not published with this error message.

```
[ERROR] [1730423319.180812735]: failed to start pipeline: Start stream failed, OB_SENSOR_DEPTH config video mode failed! try to disable ir stream try again
[ERROR] [1730423319.183086004]: Failed to start streams: Firmware stream Depth processor claimed failed!
[ INFO] [1730423319.183160533]: Image stream depth subscribed
[ INFO] [1730423319.183210862]: ====Disable frame sync====
[ERROR] [1730423319.185569220]: failed to start pipeline: Firmware stream Depth processor claimed failed! try to disable ir stream try again
[ERROR] [1730423319.187943611]: Failed to start streams: Firmware stream Depth processor claimed failed!
```

3Ddaiwei, 2024-12-05:
It seems that the output stream fails as seen in the log. It is recommended to replug the USB device and try again or provide some more log information.

obshenlanzhou, 2025-09-08:
Hi @siwook-choi Is this issue resolved? Do you need any further assistance with this issue? Thanks!

closed 2025-09-26 by obshenlanzhou