orbbec/OrbbecSDK_ROS1 #59: Device disconnected when reading  /image_raw

Rago777, 2024-11-08:
I am using a Jetson AGX Xavier with Ubuntu 18.04 and ROS Melodic. When I turn on the system, I can see that the camera is connected to the host machine:

ubuntu@ubuntu-desktop:~$ lsusb 
Bus 002 Device 002: ID 2bc5:0670 

The camera remains connected until I launch the gemini2.launch file. Despite this, an error occurs during startup:

 [ERROR] [1731062979.837715134]: Failed to setup devices: Request failed, device response with error, errorCode: 2, msg:
 
 However, when I start reading the contents of the /camera/color/image_raw topic, whether using RViz or rostopic echo, the camera immediately disconnects. At this point, lsusb also cannot detect the camera. The terminal where the ROS launch command was started shows the following:
 
 [ INFO] [1731063051.902364867]: Image stream color subscribed
[ INFO] [1731063051.902695998]: ====Enable frame sync====
[ INFO] [1731063051.902921668]: set align mode to HW
[ INFO] [1731063052.013273677]: Create color frame read thread.
[ INFO] [1731063052.013682335]: Image stream color subscribed
[ INFO] [1731063052.013947490]: pipe line already started
[ INFO] [1731063052.755705128]: deviceConnectCallback : deviceConnectCallback start
[ INFO] [1731063052.755940686]: Device disconnected
[ INFO] [1731063052.756120698]: current node serial AY3N430006T
[ INFO] [1731063052.756261642]: Device with uid 2-1-2 disconnected
[ INFO] [1731063052.756515694]: deviceDisconnectCallback : Before reset device, wait for device lock
[ INFO] [1731063052.756661342]: 2-1-2 reset device 2-1-2 notification sent
[ INFO] [1731063052.756829451]: deviceDisconnectCallback : deviceDisconnectCallback end
[ INFO] [1731063052.764826447]: resetDeviceThread: device is disconnected, reset device start
[ INFO] [1731063052.765201861]: OBCameraNode::~OBCameraNode() start
[ INFO] [1731063052.765316984]: OBCameraNode::~OBCameraNode() stop tf thread
[ INFO] [1731063052.807574156]: Color frame thread exit!
[ INFO] [1731063052.827441283]: OBCameraNode::~OBCameraNode() stop stream
[ INFO] [1731063052.829215197]: OBCameraNode::~OBCameraNode() delete rgb_buffer
[ INFO] [1731063052.829578996]: OBCameraNode::~OBCameraNode() end
[ INFO] [1731063053.324274415]: resetDeviceThread: device is disconnected, reset device
[ INFO] [1731063053.328567479]: resetDeviceThread: device is disconnected, reset device end

I am not sure whether the cause of this issue is insufficient power supply at the interface, udev permissions, or some other error. I hope to get an answer.The camera can run normally on my Ubuntu 18.04 virtual machine.

Rago777, 2024-11-08:
I am using Gemini2

obshenlanzhou, 2025-09-08:
@Rago777 Please make sure you have installed the udev rules and are using the latest commit for testing.