orbbec/OrbbecSDK_ROS1 #60: Depth image of Femto Bolt does not cover the whole color image

xaristeidou, 2024-11-16:
Migrating from Azure Kinect to Femto Bolt I am trying to find how I can enable depth image to cover the whole color image but I depth image always has this hexagon shape, as shown in the following images.

[image]
[image]
[image]

In Azure Kinect ROS driver the `/depth_to_rgb/image_raw` depth image covers the whole color image. Also there is another topic that has a very wide FOV of depth image.

I wonder because the depth sensors of Azure Kinect and Femto Bolt are identical if Femto Bolt can provide depth image that covers the whole color image.

bmegli, 2025-03-05:
> In Orbbec Viewer I enable depth image to 1024x1024 and it has wider FOV, but I have tried to stream camera through ROS1 driver modifying args of launch file of femto_bolt.launch but when I changed the values of depth_width, depth_height the depth image is not published in the topic.

Make sure to also reduce depth_fps to 15.

See the specs of Femto Bolt:
- https://www.orbbec.com/documentation/femto-bolt-hardware-specifications/

WFOV Unbinned (1024x1024) supports up to 15 fps

---------------------------------

Also make sure your device is recognized as USB 3.0 or later.

xaristeidou, 2025-03-05:
@bmegli @FishofJLlake Indeed the `depth_fps` has to be set to 15 or 5.

Also another crucial step that I just discover is that the `ir` settings have to be exactly the same as the depth. So you need also to modify them as follows:

- `depth_width=1024`
- `depth_height=1024`
- `depth_fps=15`
- `ir_width=1024`
- `ir_height=1024`
- `ir_fps=15`

This is mention in README for instructions of NFOV and WFOV:

`
Configuration of depth NFOV and WFOV modes
For the Femto Mega and Femto Bolt devices, the NFOV and WFOV modes are implemented by configuring the resolution of Depth and IR in the launch file. In launch file, depth_width、depth_height、ir_width、ir_height represents the resolution of the depth and the resolution of the IR. The frame fps and resolution of IR must be consistent with the depth. The correspondence between different modes and resolutions is as follows:

NFOV unbinned: 640 x 576.
NFOV binned: 320 x 288.
WFOV unbinned: 1024 x 1024.
WFOV binned: 512 x 512.`