orbbec/OrbbecSDK_ROS1 #71: TF `camera_accel_gyro_optical_frame` is not being published for Femto Bolt and doesn't contain timestamp

xaristeidou, 2025-02-21:
Camera: Orbbec Femto Bolt
SDK version: 1

When using `enable_accel=true` and `enable_gyro=true` and `enable_sync_output_accel_gyro=true`.

This will results to publish `/camera/gyro_accel/sample` topic which contain imu data. 

The problem is that in that topic the `camera_accel_gyro_optical_frame` the `frame_id: "camera_accel_gyro_optical_frame"` is not being publish.

Also another problem is that it doesn't contain timestamp, it is always filled as 0 secs 0 nsecs.

jjiszjj, 2025-03-10:
@xaristeidou Thank you for your feedback. 
1.Regarding the `frame_id: "camera_accel_gyro_optical_frame"` is not being published in the `/camera/gyro_accel/sample` topic
  You can try the latest Realsense version of OrbbecSDK_ROS1:https://github.com/orbbec/OrbbecSDK_ROS1/releases/tag/v2.2.4

2.It doesn't contain timestamp, it is always filled as 0 secs 0 nsecs
  You can try to update the firmware of Femto_Bolt:https://github.com/orbbec/OrbbecFirmware/releases/tag/Femto-Bolt-Firmware

dorecasan, 2025-04-25:
Hello, I face the similar problem with Femto Mega. How can I fix this problem?