orbbec/OrbbecSDK_ROS1 #72: Gemini 215 support

bmegli, 2025-03-04:
Is Gemini 215:
- https://www.orbbec.com/products/stereo-vision-camera/gemini-215/

Supported by this driver?

I can see Gemini 2, 2L listed as supported but not 215?

bmegli, 2025-03-04:
SDK side (not ROS wrapper):
-  SDK v1.10.16 added Gemini 215 support
   -  https://github.com/orbbec/OrbbecSDK/releases/tag/v1.10.16
-  SDK v2 2.1.1 added Gemini 215 support
   - https://github.com/orbbec/OrbbecSDK_v2/releases/tag/v2.1.1

bmegli, 2025-03-04:
On OrbbecSDK_ROS side
- https://github.com/orbbec/OrbbecSDK_ROS1/commit/bc475d8f59e4a8b711e122a2cd0c16f7a745a979
  - updated SDK to to 1.10.18 (commit message is wrong for 1.5.18, it can be seen in changes)

So at least SDK version used by OrbbecSDK_ROS should be compatible.

How about launchfiles/SDK_ROS code? Does it support Gemini 215?

jjiszjj, 2025-03-10:
@bmegli Hello,Thank you very much for your feedback.

1.Regarding whether OrbbecSDK_ROS1 supports Gemini 215
  The current OrbbecSDK_ROS1 does not support Gemini 215 and Gemini 210. We will consider whether to update OrbbecSDK_ROS1 to adapt to Gemini 215 and Gemini 210 in the near future.

2.commit message is wrong for 1.5.18
  Regarding https://github.com/orbbec/OrbbecSDK_ROS1/commit/bc475d8f59e4a8b711e122a2cd0c16f7a745a979 commit message is wrong for 1.5.18, this is our mistake, the correct commit message should be 1.10.18

obshenlanzhou, 2025-09-27:
The current version already supports the Gemini 210 series (including Gemini 210 and Gemini 215).
You can launch it using gemini210.launch.py.
For the camera datasheet, please refer to this link: [Gemini-210-series-Datasheet](https://d1cd332k3pgc17.cloudfront.net/wp-content/uploads/2025/08/Gemini-210-series-Datasheet-V1.4.pdf).

closed 2025-09-27 by obshenlanzhou