orbbec/OrbbecSDK_ROS1 #75: 在linux系统上运行奥比中光dabai dcw相机时候，用rviz查看彩色图片时，出现报错

15541932860, 2025-03-11:
[ERROR] [1741675587.337448118]: frame is not decoded
[ERROR] [1741675587.402934312]: Format MJPG convert to RGB888 failed
[ERROR] [1741675587.403227315]: Decode frame failed
[ERROR] [1741675587.403356236]: frame is not decoded
strmh->payload_header_hold_bytes is invalid,size:0
strmh->payload_header_hold_bytes is invalid,size:0
strmh->payload_header_hold_bytes is invalid,size:0

jjiszjj, 2025-03-12:
@15541932860 Hello,Thank you very much for your feedback.
The color stream video frame is lost, causing some video frames to fail to decode.

jjiszjj, 2025-03-14:
@15541932860 Camera frame loss is currently unavoidable. You can try to check the USB connection cable and port.