orbbec/OrbbecSDK_ROS1 #76: gemini定制模块相机 升级 使 虚拟机崩溃

metadingding, 2025-03-14:
orbbec合作方的百度网盘https://pan.baidu.com/e/1inZOYXYYeba_rTAmhnqQZA，提取码gmax
下载“Gemini MAX深度相机>ROS1>ROS1虚拟机”
ubuntu20.04/debian虚拟机，ros版本noetic，系统里预安装的是1.8.1的libOrbbecSDK.so

虚拟机启动后，相机是正常运行的。但没有新版本launch中的关闭红外激光的功能。

为获取此功能，用github包在orbbec_ws中catkin_make升级到1.10.18，然后launch gemini.launch，会在切换话题（比如rqt_image_view中）或退出相机时，使虚拟机崩溃死机！

我切换到1.10.17，1.10.16等，都会如此。
但降低版本，从1.8.1降到1.5.7，就没有任何问题

是激光投影的开关功能导致系统崩溃？还是什么问题？
麻烦帮忙给看一下

metadingding, 2025-03-14:
[image]

obshenlanzhou, 2025-09-04:
Hi @metadingding Can you  provide the specific camera model you are using? And system version? Branch? This will help us resolve the issue more quickly.