orbbec/OrbbecSDK_ROS1 #79: multi camera.launch 启动失败

kourner, 2025-04-17:
已安装 udev rules。

[image]

[image]

jjiszjj, 2025-04-17:
@kourner You can run this command `rosrun orbbec_camera list_devices_node` to get the camera's `usb_port`, and then fill the obtained `usb_port` into `multi_camera.launch`

kourner, 2025-04-17:
> [@kourner](https://github.com/kourner) You can run this command `rosrun orbbec_camera list_devices_node` to get the camera's `usb_port`, and then fill the obtained `usb_port` into `multi_camera.launch`

已查询并设置过了

[image]

[image]

jjiszjj, 2025-04-17:
> > [@kourner](https://github.com/kourner) You can run this command `rosrun orbbec_camera list_devices_node` to get the camera's `usb_port`, and then fill the obtained `usb_port` into `multi_camera.launch`
> 
> 已查询并设置过了
> 
> [image]
> 
> [image]

@kourner You can try replacing the `rosrun orbbec_camera list_ob_devices.sh` command with `rosrun orbbec_camera list_devices_node`

kourner, 2025-04-17:
> [@kourner](https://github.com/kourner) You can try replacing the `rosrun orbbec_camera list_ob_devices.sh` command with `rosrun orbbec_camera list_devices_node`

感谢，使用rosrun orbbec_camera list_devices_node查询的usb port好像是可以了。但又出现了新的问题，还请帮忙再看一下。

[image]

[image]

jjiszjj, 2025-04-24:
@kourner Hi,I found that the camera you are using is DaBai DCW, you need to change gemini_330_series to dabai_dcw

[image]

closed 2025-05-06 by kourner