orbbec/OrbbecSDK_ROS1 #82: Component 8 not found!

l-j-h-666, 2025-05-23:
Hi, I tried to use orbbec gemini2 for VIO, thus RGBD and IMU data is needed. 
Firstly, an issue similar to <a href=https://github.com/orbbec/OrbbecSDK_ROS1/issues/71>issue71</a> is occured: `camera_accel_gyro_optical_frame` is not published in TF. Following the suggestions in issue71 and Orbbec Firmware, OrbbecSDK_ROS1 interface is upgraded to v2.2.4 and gemini2 firmware to v1.4.92. After launching `roslaunch orbbec_camera gemini2.launch` the problem **periodically** comes: 

各位好，我们在用gemini2做VIO，需要RGBD和IMU数据。
首先我们碰到了<a href=https://github.com/orbbec/OrbbecSDK_ROS1/issues/71>issue71</a>类似的问题：`camera_accel_gyro_optical_frame`在tf中没有发布。然后我们根据建议使用了v2.2.4的ROS1接口并且升级gemini2固件到1.4.92版本，在`roslaunch orbbec_camera gemini2.launch`后**周期性**报错如下：
```
[INFO] [1747976550.500757766]: Connecting to the default device
[INFO] [1747976550.556185516]: deviceConnectCallback : selectDevice end
[INFO] [1747976550.663409098]: hdr_index1_laser_control_ 1 hdr_index1_depth_exposure_ 1 hdr_index1_depth_gain_ 16 hdr_index1_ir_brightness_ 20 hdr_index1_ir_ae_max_exposure_ 2000

[INFO] [1747976550.663500559]: hdr_index0_laser_control_ 1 hdr_index0_depth_exposure_ 7500 hdr_index0_depth_gain_ 16 hdr_index0_ir_brightness_ 60 hdr_index0_ir_ae_max_exposure_ 10000

[INFO] [1747976550.663542898]: laser_index1_laser_control_ 0 laser_index1_depth_exposure_ 3000 laser_index1_depth_gain_ 16 laser_index1_ir_brightness_ 60 laser_index1_ir_ae_max_exposure_ 7000

[INFO] [1747976550.663576468]: laser_index0_laser_control_ 1 laser_index0_depth_exposure_ 3000 laser_index0_depth_gain_ 16 laser_index0_ir_brightness_ 60 laser_index0_ir_ae_max_exposure_ 17000

[INFO] [1747976550.663633623]: current time domain:device
[INFO] [1747976550.664067727]: enable_sync_host_time:1
[ERROR] [1747976550.775959254]: Failed to initialize device Component 8 not found!
[INFO] [1747976550.776170498]: deviceConnectCallback : deviceConnectCallback end
```
The main problem is `Failed to initialize device Component 8 not found!`. 
可以注意到主要问题为`Failed to initialize device Component 8 not found!`。

More details are displayed using `loglevel:="debug"`:
通过`loglevel:="debug"`可以看到更多的报错细节：
```
[05/23 13:15:36.017197][debug][13372][VideoSensor.cpp:308] Backend stream profile {type: Video, format: Y16, width: 1280, height: 800, fps: 10}
[05/23 13:15:36.017239][debug][13372][VideoSensor.cpp:308] Backend stream profile {type: Video, format: Y16, width: 1280, height: 800, fps: 5}
[05/23 13:15:36.019058][debug][13372][VideoSensor.cpp:38] VideoSensor created @Depth
[05/23 13:15:36.039986][debug][13372][PropertyServer.cpp:162] Property 85 set to 0|0
[05/23 13:15:36.040022][warning][13372][ObException.cpp:8] Component 8 not found!
[05/23 13:15:36.040085][debug][13372][VideoSensor.cpp:49] VideoSensor destroyed @Depth
[05/23 13:15:36.040119][info][13372][SensorBase.cpp:34] SensorBase is destroyed
[05/23 13:15:36.040142][warning][13372][DeviceBase.cpp:268] Execute failure! A libobsensor_exception has occurred!
 - where: DeviceBase.cpp(268):getComponent
 - msg: Component 8 not found!
 - type: N11libobsensor23invalid_value_exceptionE
[05/23 13:15:36.040244][debug][13372][FilterDecorator.cpp:299] Filter@PointCloudFilter have been reset
[05/23 13:15:36.040253][debug][13372][FilterDecorator.cpp:284] Filter PointCloudFilter destroyed
[05/23 13:15:36.040269][debug][13372][FilterDecorator.cpp:299] Filter@PointCloudFilter have been reset
[05/23 13:15:36.040273][debug][13372][FilterDecorator.cpp:284] Filter PointCloudFilter destroyed
[05/23 13:15:36.040300][debug][13372][FilterDecorator.cpp:299] Filter@FormatConverter have been reset
[05/23 13:15:36.040304][debug][13372][FilterDecorator.cpp:284] Filter FormatConverter destroyed
[ERROR] [1747977336.040393562]: Failed to initialize device Component 8 not found!
[INFO] [1747977336.040819176]: resetDeviceThread: device is disconnected, reset device start
[INFO] [1747977336.040851306]: resetDeviceThread: device is disconnected, reset device
```

请问应该如何解决？
Any solutions?

jjiszjj, 2025-05-23:
@l-j-h-666 You can check whether there are environment variable configurations for multiple ROS workspaces in the current terminal environment.
`echo $ROS_PACKAGE_PATH`

l-j-h-666, 2025-05-23:
> [@l-j-h-666](https://github.com/l-j-h-666) You can check whether there are environment variable configurations for multiple ROS workspaces in the current terminal environment. `echo $ROS_PACKAGE_PATH`

Thank you, and EnvVar path for orbbec_ROS1 workspace DOES exist in the current terminal by echoing `ROS_PACKAGE_PATH`, in my case it is `/home/jetson/radar_demo/src:/home/jetson/clouder_ws/src:/home/jetson/camera_ws/src:/opt/ros/noetic/share` where `camera_ws` is the corresponding workspace.

LianLian0915, 2025-08-14:
你好，所以这个问题解决了吗？我这边使用335Le的时候也出现了类似的问题，不过报的是Failed to initialize device Component 12 not found!

LianLian0915, 2025-08-16:
> 你好，所以这个问题解决了吗？我这边使用335Le的时候也出现了类似的问题，不过报的是Failed to initialize device Component 12 not found!

将版本降至2.4.2以后问题解决了

LianLian0915, 2025-08-25:
> > > 你好，所以这个问题解决了吗？我这边使用335Le的时候也出现了类似的问题，不过报的是Failed to initialize device Component 12 not found!
> > 
> > 
> > 将版本降至2.4.2以后问题解决了
> 
> 摄像头的固件怎么降级啊？

我并没有降级固件 只是将这个工程的版本降低到2.4.2了

sun-xm, 2025-09-18:
我在windows上使用astra2时遇到了同样的“component 8 not found"问题。拷贝奥比SDK中bin目录下的extensions文件夹到执行路径中问题得以解决。可能是OrbbecSDK.dll动态加载了某个extensions文件夹中的dll没有成功。具体原因没有深究。