orbbec/OrbbecSDK_ROS1 #84: [Bug] NoiseRemovalFilter#1 config missing – fails device setup and blocks powerline frequency setting

HTLife, 2025-06-10:
## Environment:
- Camera: Orbbec Gemini 336
- Host: Ubuntu 20.04 on NVIDIA Jetson Orin Nano (JetPack 5.1.3)
- SDK Version: OrbbecSDK_v2 Release v2.4.3
- Firmware: Gemini 330 series (v1.4.60)
- ROS Wrapper: OrbbecSDK_ROS1 (branch: v2-main, commit: a4a87b197c3f56ccc10d3690b58fafdc879a75d0)

## Steps to Reproduce

1. Launch ROS node using: `roslaunch orbbec_camera gemini_330_series.launch`
2. Observe [ERROR] message 

```
[ERROR] [1749542154.126314838]: Failed to setup devices: Filter@FrameProcessor: config item NoiseRemovalFilter#1 doesn't exist
```
This exception originates from src/ros_setup.cpp in the setupDevices() function.

As a result, the powerline frequency configuration (`<arg name="color_powerline_freq" default="50hz"/>`) has no effect. 

The RGB camera continues to exhibit flickering under 50Hz indoor lighting conditions.

## Request
Please check whether the default device pipeline config is missing the `NoiseRemovalFilter#1` or if it's incorrectly referenced. 
Also, confirm if this blocks `OB_PROP_COLOR_POWER_LINE_FREQUENCY_INT` setup, and suggest a workaround or patch to allow frequency adjustment.

obshenlanzhou, 2025-08-21:
I tested on my side and did not encounter this issue. However, this error does affect the `color_powerline_freq` parameter setting. You can try the following method：

1. You can try the latest Realsense version of OrbbecSDK_ROS1 to resolve the error.

2. You can try to update the firmware：https://www.orbbec.com/docs/g330-firmware-release/

obshenlanzhou, 2025-09-04:
Hi @HTLife Is this issue resolved? Do you require further assistance with this case？

closed 2025-10-30 by obshenlanzhou