orbbec/OrbbecSDK_ROS1 #85: Failed to start stream, Extension info IspNeedVer not found!

feixyz10, 2025-06-10:
Env: ubuntu 20.04, ros noetic, x86-64
Device: Gemini 335L.

Works well if I use Orbbec Viewer, but problem is encountered if I use ROS.

In one terminal, I launch camere node: `roslaunch orbbec_camera gemini_330_series.launch`

Once some other node try to subscribe camera-related topics, e.g. `rostopic echo /camera/color/camera_info`, following error message will be shown:

```
Failed to start pipeline: Extension info IspNeedVer not found! try to disable ir stream try again
Failed to start stream: Extension info IspNeedVer not found!
```

feixyz10, 2025-06-11:
Update: v2-main branch will give above error. main branch is totally ok.

jjiszjj, 2025-06-11:
The v2-main branch currently does not support some old firmware. You can try upgrading the firmware.https://www.orbbec.com/docs/g330-firmware-release/

closed 2025-08-12 by winf